adafruit/Adafruit_CircuitPython_ADS1x15 #11: ADC conversion timing issue for ADS1115

caternuson, 2018-07-02:
It takes two reads to get the current value using an ADS1115.
```python
>>> import board, busio
>>> from adafruit_ads1x15.single_ended import ADS1115
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> adc = ADS1115(i2c)
>>> # A0 initially connected to VDD
>>> adc[0].value
26391
>>> # A0 connected to GND
>>> adc[0].value
26391
>>> adc[0].value
0
```
This timing kludge (ported from Arduino version of lib) isn't working:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/blob/master/adafruit_ads1x15/adafruit_ads1x15.py#L197
Also see general issue with timing resolution:
https://github.com/adafruit/circuitpython/issues/429

Need to either increase the delay, or better yet, actually poll and wait for conversion to complete.

ladyada, 2018-07-03:
lets just poll, wanna add that?

caternuson, 2018-07-03:
Works. Was thinking of adding a timeout to the poll loop. But for now, it's just a crazy simple loop.

PR #12 (2018-07-10): Add wait for conversion
This adds a wait for conversion polling loop to fix #11

However, also refactored a lot of the boiler plate register read / write code into private funcs.
closed 2018-07-10 by caternuson