adafruit/Adafruit_CircuitPython_ADS1x15 #26: Support continuous mode

caternuson, 2019-03-04:
The current wait-for-conversion loop will block forever in continuous mode.

jmn98, 2019-03-06:
I have been trying to use continuous mode and have only seen erroneous readings from the ADC. See code below, possibly an implementation error but i suspect my issue is related to the library. 

`import numpy as np
import time
import board
import busio
import adafruit_ads1x15.ads1015 as ADS

i2c = busio.I2C(board.SCL, board.SDA)
ads = ADS.ADS1015(i2c)
np.set_printoptions(suppress = True)

ads.read(0, is_differential = True)

ads.mode = 0 

ads.data_rate = 3300
ads.gain = 2/3

voltage = ads.get_last_result()`

Results were sporadic and not alligned with the reference voltage I was measuring. My aim was to measure the differential between channel 0 and channel 1. 

Cheers,
J

caternuson, 2019-03-07:
These lines:
```python
ads.mode = 0
ads.data_rate = 3300
ads.gain = 2/3
```
only set member variables and don't actually get sent to the ADS until a call to `_read`, which isn't happening in your code.

caternuson, 2019-05-03:
Oops. Looks like I accidentally deleted / closed this doing some house keeping? Not sure. Wasn't intentional. I guess if anyone actually still wants this, respond here and I'll see if I can bring it back.

chrisjamesfell, 2019-05-04:
Thanks Carter. I'd really like to see Python support for Continuous mode on this device. I think the ADS1x15 is an excellent option for sensor data aquisition on the RPi, but the range of applications is limited if the driver is limited to using single-shot mode, as I believe the AnalogIn function is.

The limitation at present is that if you want to sample for more than 1/8th of a second (or even shorter for higher sample rates), you need to run many single-shot reads onto a loop. I believe looping single-shot reads leads to very patchy and inefficient sampling, due to the delays associated with (i) code execution, and more importantly (ii) the ADS switching to low-power mode after each read, and then having to ramp the power back up for the next read.

This is literally my first ever GitHib post, so I hope I'm doing it right. I love how somebody at the other end of the internet is working to make all this cool stuff work :)

Regards, Chris.

caternuson, 2019-05-04:
OK, I was able to reopen #28 without much trouble. Still not sure when/how/why it get closed.

caternuson, 2019-05-09:
Fixed by #28

closed 2019-05-09 by caternuson