adafruit/Adafruit_CircuitPython_ADS1x15 #36: Compatibility w/ TCA9548A

VicerExciser, 2019-07-02:
Is it possible to multiplex these ADCs using this circuitpython implementation? I'm trying to leverage the https://github.com/adafruit/Adafruit_CircuitPython_TCA9548A library for reading more than 4 chained ADS1015s on my Raspberry Pi's I2C bus, but the suggested method for substituting the ADS instance's 'i2c' attribute with a TCA Channel object is not working; the ADS1x15.__init__ function's attempt to locate address 0x48 on my bus crashes my program, as i2cdetect only recognizes the TCA mux address (0x70). Any help would be appreciated.

caternuson, 2019-07-02:
Should work. Are you using the Adafruit ADS1015 breakout or something else?

VicerExciser, 2019-07-02:
Yes, I am. Quite a few of them. Unfortunately there is only one example for using the circuitpython TCA9548A class and it's for multiplexing dedicated I2C analog sensor breakouts in a sort of plug-and-play fashion. The ADCs have to do a little multiplexing of their own so perhaps that's where the complication lies? Looks like the ADS1x15 base class performs stricter checking of its presence on the bus than most other Adafruit circuitpython modules.

VicerExciser, 2019-07-02:
Ah, it's because the TCA module will only let me read from a different address than its own default 0x70, which as I said is the only address visible on the bus. And the ADS base class uses I2CDevice from adafruit_bus_device.i2c_device, which requires a findable address. This is starting to sound more appropriate for the TCA9548A repo's issue tracker...

caternuson, 2019-07-02:
You'll always see the address of the TCA itself on the bus. But if you've configured it and set its output channel to the one the ADS is on, then you should also see that address:
https://learn.adafruit.com/adafruit-4-channel-adc-breakouts/assembly-and-wiring#i2c-addressing-2-12
So, for example, 0x70 (TCA) and 0x48 (ADS).

It seems odd that you are not seeing the ADS address.

VicerExciser, 2019-07-02:
So maybe it’s an issue with setting an output channel on the TCA then? Maybe that needs to be done before trying to instantiate an ADS object, but I would assume that it would be set to channel 0 by default...right?

For my mux test circuit I just have 5 ADS breakouts connected to TCA channels 0-4, such that channel 0 -> 0x48, 1 -> 0x49, 2 -> 0x4a, 3 -> 0x4b, and 4 -> 0x48. By default, none of those assignable ADS addresses are visible, only 0x70. So perhaps ensuring that each respective channel is enabled/open before creating the corresponding ADS instances might work. Will test this once I’m back in the lab.

caternuson, 2019-07-02:
Only one TCA output channel is active at a time. So an I2C scan will only see the ADS(s) on that channel. The CircuitPython TCA driver takes care of changing the channel for you. But if you are trying to manually scan the I2C bus, then you'll need some way to also manually switch the TCA output channels.

This might be better handled in the forums:
https://forums.adafruit.com/
where you can post a photo of your setup so we can check for any wiring issues also.

VicerExciser, 2019-07-02:
Turned out to be a hardware error (of course...), nothing more than a bad wire that wasn't providing power to one of the ADS's. The others' addresses were being recognized on the bus after all but the program would crash once a read was attempted for that dead ADS. Issue resolved and lesson learned.

closed 2019-07-02 by VicerExciser