adafruit/Adafruit_CircuitPython_ADS1x15 #38: Strange output

laloshifrin, 2019-08-22:
I'm using Raspberry Pi 3B+ with Raspbian Stretch.
Using this code:

`import time
import board
import busio
#import adafruit_ads1x15.ads1015 as ADS
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.analog_in import AnalogIn

#Create the I2C bus
i2c = busio.I2C(board.SCL, board.SDA)

#Create the ADS object
#ads = ADS.ADS1015(i2c)
ads = ADS.ADS1115(i2c)

#Create a sinlge ended channel on Pin 0
#Max counts for ADS1015 = 2047
#ADS1115 = 32767
chan = AnalogIn(ads, ADS.P0)

#The ADS1015 and ADS1115 both have the same gain options.
#
#GAIN    RANGE (V)
#----    ---------
#2/3    +/- 6.144
#1    +/- 4.096
#2    +/- 2.048
#4    +/- 1.024
#8    +/- 0.512
#16    +/- 0.256
#
gains = (2/3, 1, 2, 4, 8, 16)

while True:
        ads.gain = gains[0]
        print('{:5} {:5.3f}'.format(chan.value, chan.voltage), end='')
        for gain in gains[1:]:
            ads.gain = gain
            print(' | {:5} {:5.3f}'.format(chan.value, chan.voltage), end='')
        print()
        time.sleep(0.5)`

I get following readings:

 1955 0.495 |  4249 0.536 |  4814 0.270 |  8471 0.264 |  1465 0.017 |  2177 0.017
 1927 0.489 |  4193 0.531 |  4770 0.268 |  8467 0.264 |  1004 0.010 |  1221 0.012
 1913 0.494 |  4191 0.533 |  4761 0.269 |  8453 0.264 |  1041 0.010 |  1219 0.010
 1926 0.491 |  4186 0.534 |  4760 0.270 |  8448 0.264 |  1076 0.010 |  1259 0.012
 1936 0.491 |  4212 0.532 |  4772 0.270 |  8450 0.265 |  1010 0.012 |  1305 0.010
 1932 0.493 |  4212 0.532 |  4799 0.269 |  8458 0.264 |   988 0.012 |  1211 0.013
 1916 0.496 |  4009 0.530 |  4781 0.270 |  8445 0.265 |  1023 0.011 |  1383 0.011
 1931 0.494 |  4198 0.534 |  4773 0.270 |  8468 0.265 |  1000 0.011 |  1505 0.009
 1931 0.494 |  4200 0.534 |  4768 0.271 |  8445 0.265 |  1056 0.010 |  1421 0.011
 1946 0.491 |  4214 0.532 |  4781 0.271 |  8453 0.266 |  1016 0.010 |  1392 0.011
 1944 0.493 |  4214 0.533 |  4782 0.270 |  8463 0.264 |  1040 0.010 |  1305 0.010
 1930 0.494 |  4199 0.535 |  4767 0.270 |  8454 0.264 |  1046 0.010 |  1255 0.010
 1941 0.491 |  4210 0.534 |  4767 0.271 |  8450 0.265 |  1008 0.011 |  1299 0.012
 1934 0.493 |  4198 0.532 |  4797 0.270 |  8468 0.265 |   963 0.010 |  1537 0.010
 1941 0.495 |  4203 0.534 |  4797 0.270 |  8482 0.264 |  1067 0.010 |  1422 0.010
 1926 0.494 |  4197 0.535 |  4777 0.271 |  8472 0.264 |   999 0.010 |  1130 0.013
from a probe which gives a few millivolts on channel0

and these:
 3016 0.573 |  4638 0.577 |  4484 0.278 |  8656 0.270 |    42 0.001 |    84 0.001
 3037 0.574 |  4628 0.577 |  4479 0.278 |  8663 0.271 |    42 0.001 |    86 0.000
 3016 0.577 |  4618 0.578 |  4469 0.279 |  8656 0.270 |    42 0.000 |    85 0.000
 3014 0.576 |  4614 0.579 |  4462 0.279 |  8639 0.270 |    43 0.001 |    45 0.001
 3030 0.573 |  4630 0.578 |  4465 0.279 |  8642 0.271 |    42 0.001 |    45 0.001
 3032 0.575 |  4630 0.577 |  4482 0.278 |  8656 0.270 |    43 0.000 |    85 0.001
 3018 0.578 |  4611 0.577 |  4482 0.279 |  8654 0.270 |    42 0.001 |    84 0.001
 3023 0.578 |  4620 0.579 |  4471 0.279 |  8658 0.270 |    61 0.001 |    86 0.001
 3024 0.575 |  4617 0.580 |  4458 0.280 |  8642 0.271 |    42 0.001 |    45 0.001
 3040 0.574 |  4633 0.578 |  4463 0.279 |  8649 0.271 |    42 0.001 |    84 0.001
 3037 0.575 |  4630 0.577 |  4485 0.278 |  8668 0.270 |    41 0.000 |    85 0.001
 3018 0.576 |  4620 0.579 |  4472 0.278 |  8651 0.270 |    42 0.001 |    85 0.000
 3026 0.575 |  4621 0.579 |  4463 0.279 |  8638 0.271 |    42 0.001 |    45 0.001
 3039 0.572 |  4633 0.578 |  4462 0.279 |  8642 0.271 |    44 0.001 |    44 0.001
on channel1 without anithyng connected.

Those readings don't seem to me congruent because the growing gain should give growing values. Or it shouldn't? I'm a little bit confused, can anybody help?
Thank you :-)

caternuson, 2019-08-22:
Can you provide more details about what is actually connect to the ADC channels and how.

I can not recreate this. I wired up an ADS1115 to Pi Zero W and setup a 10k trim pot to voltage divide VCC (3.3V) wired in to channel 0. Set the divider to be about ~0.5V and ran your sketch above. Here's the output:
```console
pi@pizerow:~ $ python3 ads_test.py 
 2642 0.495 |  3964 0.495 |  7927 0.495 | 15856 0.495 | 31710 0.495 | 32767 0.256
 2642 0.495 |  3963 0.495 |  7928 0.496 | 15855 0.495 | 31710 0.495 | 32767 0.256
 2642 0.495 |  3963 0.496 |  7927 0.495 | 15854 0.495 | 31707 0.495 | 32767 0.256
 2642 0.495 |  3963 0.496 |  7927 0.496 | 15856 0.496 | 31708 0.495 | 32767 0.256
 2642 0.495 |  3963 0.496 |  7928 0.496 | 15857 0.495 | 31709 0.495 | 32767 0.256
 2642 0.496 |  3964 0.496 |  7928 0.496 | 15855 0.495 | 31710 0.496 | 32767 0.256
 2643 0.496 |  3964 0.495 |  7927 0.495 | 15855 0.496 | 31709 0.495 | 32767 0.256
 2643 0.496 |  3965 0.496 |  7927 0.495 | 15856 0.496 | 31710 0.495 | 32767 0.256
```
The raw values increase as expected with the last gain setting being saturated, which is also expected for ~0.5V input. The voltage values (except for the saturated case) are all about the same, since the voltage math takes into account the gain setting.

laloshifrin, 2019-08-22:
I have a PH probe directly connected with BNC connector on channel 0 and nothing connected to channel 1
I also tested channel 2 connecting VDD to it:
```
17450 3.270 | 26161 3.270 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17449 3.272 | 26180 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17451 3.272 | 26179 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17453 3.272 | 26182 3.273 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17453 3.272 | 26180 3.273 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17451 3.272 | 26177 3.273 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17453 3.274 | 26178 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17451 3.271 | 26177 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17453 3.271 | 26176 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17452 3.272 | 26176 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17451 3.271 | 26178 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17451 3.272 | 26178 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17450 3.271 | 26179 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17451 3.274 | 26177 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
17450 3.272 | 26178 3.272 | 32767 2.048 | 32767 1.024 | 32767 0.512 | 32767 0.256
```
and connecting GND to it:
```
  -23 -0.005 |   -32 -0.004 |   -66 -0.004 |  -124 -0.004 |  -268 -0.004 |  -503 -0.004
  -21 -0.004 |   -33 -0.004 |   -62 -0.004 |  -117 -0.004 |  -250 -0.004 |  -457 -0.004
  -21 -0.004 |   -30 -0.004 |   -63 -0.004 |  -120 -0.004 |  -242 -0.004 |  -466 -0.004
  -22 -0.004 |   -31 -0.004 |   -60 -0.004 |  -116 -0.004 |  -255 -0.004 |  -453 -0.004
  -20 -0.004 |   -32 -0.004 |   -62 -0.004 |  -115 -0.004 |  -251 -0.004 |  -460 -0.004
  -20 -0.004 |   -31 -0.004 |   -61 -0.004 |  -111 -0.004 |  -243 -0.004 |  -460 -0.004
  -20 -0.004 |   -32 -0.004 |   -62 -0.004 |  -114 -0.004 |  -247 -0.004 |  -475 -0.004
  -22 -0.004 |   -32 -0.004 |   -61 -0.004 |  -115 -0.004 |  -253 -0.004 |  -468 -0.004
  -24 -0.004 |   -31 -0.004 |   -61 -0.004 |  -121 -0.004 |  -249 -0.004 |  -476 -0.004
  -19 -0.004 |   -32 -0.004 |   -63 -0.004 |  -115 -0.003 |  -252 -0.004 |  -467 -0.004
  -20 -0.004 |   -33 -0.004 |   -61 -0.004 |  -118 -0.004 |  -258 -0.004 |  -461 -0.004
  -21 -0.004 |   -30 -0.004 |   -65 -0.004 |  -115 -0.004 |  -246 -0.004 |  -480 -0.004
  -21 -0.005 |   -30 -0.004 |   -61 -0.004 |  -121 -0.004 |  -254 -0.004 |  -460 -0.004
  -22 -0.004 |   -30 -0.004 |   -62 -0.004 |  -116 -0.004 |  -246 -0.004 |  -451 -0.004
  -21 -0.004 |   -29 -0.004 |   -60 -0.004 |  -116 -0.004 |  -252 -0.004 |  -464 -0.004
```
those seem more reliable but what's going on with that probe?

caternuson, 2019-08-22:
Those do look more reasonable.

Unknown what is happening but this sounds more like a hardware issue. You can try posting in the forums:
https://forums.adafruit.com/

laloshifrin, 2019-08-23:
what readings do you get if you execute the script on a channel with nothing connected?

caternuson, 2019-08-23:
In that case I do seem to get odd results similar to yours:
```python
pi@pizerow:~ $ python3 ads_test.py 
 3119 0.590 |  4740 0.589 |  4516 0.285 |  8786 0.274 |    62 0.001 |    45 0.001
 3111 0.597 |  4706 0.586 |  4566 0.284 |  8779 0.275 |    42 0.001 |   122 0.000
 3164 0.584 |  4726 0.592 |  4528 0.282 |  8802 0.275 |    42 0.001 |    44 0.000
 3103 0.591 |  4735 0.587 |  4529 0.285 |  8776 0.275 |    62 0.000 |    84 0.001
 3160 0.589 |  4694 0.591 |  4560 0.282 |  8801 0.275 |    43 0.001 |   124 0.001
 3180 0.586 |  4703 0.593 |  4547 0.281 |  8810 0.275 |    42 0.001 |    45 0.001
 3099 0.595 |  4722 0.586 |  4532 0.285 |  8777 0.275 |    62 0.000 |    44 0.001
```

laloshifrin, 2019-08-23:
You are right. It's a hardware problem related to impedance. Thanks for your time

closed 2019-08-23 by laloshifrin