adafruit/Adafruit_CircuitPython_ADS1x15 #39: Two examples need renaming to match lib name

kattni, 2019-09-30:
Two examples begin with `ads1015_` and `ads1115_`. They should each begin with `ads1x15_ads1015_` and `ads1x15_ads1115_` to match the CIrcuitPython library standards and make it clearer in the example bundle which library they are associated with.

razinc, 2019-10-12:
Hi @kattni. I would like to take this but I don't have a clear picture of the issue. One of the requirements is changing the word `ads1015_` to `ads1x15_ads1015_` & `ads1115_` to `ads1x15_ads1115_` in `docs/examples.rst`. Does it mean I have to put the prefix `ads1x15` to each file in this repo?

kattni, 2019-10-12:
Hello, @razinc. 

The prefix `ads1x15_` needs to be added to two of the example files in this repo (the two files that include `simpletest` in the name), the two other examples already have that as the beginning of their name (the two files that do not include `simpletest` in their name). 

Once the filenames are changed, the `docs/examples.rst` file needs the file names in it updated, as it includes both of the files that need to be renamed.

razinc, 2019-10-12:
Okay. Let me take this, please.

dherrada, 2020-01-13:
Closed. Fixed in #41

closed 2020-01-13 by dherrada