adafruit/Adafruit_CircuitPython_ADS1x15 #40: Can cause crashes when used with multiple threads at once.

hamishmb, 2019-10-04:
After trying to diagnose an intermittent crash in some software, I finally came to the realisation that multiple threads were querying the ADS at once. Due to the way the code works, it currently isn't thread safe and can deadlock waiting for a response from the ADS.

I've made a solution in the software using threading.RLock, but I was wondering if anyone would be interested in fixing this upstream in this code as well. If so, I'll fork, make my modifications, and submit a pull request.

Hamish

ladyada, 2019-10-04:
ideally we would add peripheral locking support to `blinka` - we do in the microcontroller - then this line would lock the peripheral
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/blob/master/adafruit_ads1x15/ads1x15.py#L198
let me know if you're interested in doing that, it would solve the problem over all for any sensor then :)

hamishmb, 2019-10-04:
That sounds like a good idea.

I'm not familiar with blinka, but by the sounds of it that would be a much more helpful fix. I don't have any ADS/other adafruit devices to hand to make sure it works, so would someone else be able to test it works with real hardware for me?

hamishmb, 2019-10-04:
Looks like blinka is a pretty complex project. Are there any docs that could help me get an idea of what is where?

ladyada, 2019-10-04:
how are you testing this PR?

hamishmb, 2019-10-04:
I tested the software modification I made to my project with real hardware, but I don't have access to it any more - it's part of an operational system. I may be able to write a test program that exhibits the issue for a colleague, who can then check that the fix works.

hamishmb, 2019-10-13:
I've sorted this out with my colleague - I can borrow some hardware from them to work on and test this fix. I'll start work on this next week and hopefully will have something ready at the weekend.

hamishmb, 2019-10-13:
Do you have a unit test framework for Blinka? I'm wondering how to make an automated test for this fix. I can write a test program, but I assume it would be better to integrate it into the unit tests.

ladyada, 2019-10-13:
we do, however nothing finds bugs like real hardware uses :)

hamishmb, 2019-10-31:
Apologies for the delay.

I have added my fix into the repository at https://github.com/hamishmb/Adafruit_Blinka. I have also written a test program, which is attached. I'm having issues getting it to crash for some reason, but it works fine with the modified, thread-safe version.

ladyada, 2019-10-31:
hiya yes please submit PR's to Blinka, our maintainer will look & test :)

FoamyGuy, 2020-02-25:
Looks like this was handled with PR #170  from the Blinka project. I'm going to close the issue here.

closed 2020-02-25 by FoamyGuy