adafruit/Adafruit_CircuitPython_ADS1x15 #43: not getting 1600sps in ads1015

nayanagg, 2019-10-23:


caternuson, 2019-10-23:
Obtaining the fastest sampling rate may not be possible. See here for lots of details and discussion:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/issues/27
The best this library supports is demonstrated in this example:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/blob/master/examples/ads1x15_fast_read.py

nayanagg, 2019-10-23:
[image]
After using fast_read.py regardless of what sample i'm giving the result is same.

caternuson, 2019-10-23:
The general issue is syncing with the conversion completion in continuous mode. There are two general ways this can be done:

1. Query the status register for conversion complete
2. Query the conversion complete pin, ideally tie this to an interrupt handler

Approach 2 is not viable since CircuitPython does not support interrupts. Approach 1 has the issue of slowing everything down since the query requires an I2C transaction. Checking this status bit is done for regular reads:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/blob/master/adafruit_ads1x15/ads1x15.py#L166
but not in the special fast read mode:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/blob/master/adafruit_ads1x15/ads1x15.py#L154

You'll have to deal with the conversion complete syncing somehow:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/issues/27#issuecomment-494598221
Note that the final approach that was taken is generally very simple and did not end up using the ALRT pin:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/issues/27#issuecomment-497406750

ahmetuludag, 2019-10-25:
I followed links and there is great effort and insight, thank you @caternuson. In a project; I'm using ADS1115 and needed to have a stable 475 sample rate. As you already pointed out, fast_read example tries to read max and does not obey settings. There is also this library: https://pypi.org/project/RPi.GPIO/ and its comment says that for real-time performance don't use raspberry, arduino would be better.

So my question; is 475 sample rate in a stable manner (+-1 could be OK) possible on Raspberry Pi? (At least theoretically? Assume that I found a way to use interrupts)

caternuson, 2019-10-31:
Unknown. But worth trying. Would be interesting to see what you can work out.

kattni, 2020-03-30:
@caternuson Is this resolved?

caternuson, 2020-03-30:
We can close for now.

Not really resolved, but I don't think there's any easy to do this with the current state of CP (i.e., no interrupts). The discussions in the threads linked above document where things are at.

closed 2020-03-30 by caternuson