adafruit/Adafruit_CircuitPython_ADS1x15 #48: TypeError: an integer is required (got type bytearray)

Grandpa-G, 2020-01-03:
I am trying to get this library running on Raspberry Pi under buster OS. The following is both the code and the error. Please advise what to do to correct? I can use the old deprecated version and it works so I know the hardware works and the environment should be setup correctly. I have also installed blinka library.

`import board
import busio
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.analog_in import AnalogIn
 
i2c = busio.I2C(board.SCL, board.SDA)
 
 
ads = ADS.ADS1115(i2c, address=0x4a)
 
chan = AnalogIn(ads, ADS.P0)
 
print(chan.value, chan.voltage)
------------------------------------ output
sudo python3 temp.py
 
Traceback (most recent call last):
  File "temp.py", line 13, in <module>
    print(chan.value, chan.voltage)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ads1x15/analog_in.py", line 74, in value
    is_differential=self.is_differential) << (16 - self._ads.bits)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ads1x15/ads1x15.py", line 137, in read
    return self._read(pin)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ads1x15/ads1x15.py", line 166, in _read
    while not self._conversion_complete():
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ads1x15/ads1x15.py", line 176, in _conversion_complete
    return self._read_register(_ADS1X15_POINTER_CONFIG) & 0x8000
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ads1x15/ads1x15.py", line 202, in _read_register
    i2c.write_then_readinto(bytearray([reg]), self.buf, in_end=2)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 136, in write_then_readinto
    in_start=in_start, in_end=in_end)
  File "/usr/local/lib/python3.7/dist-packages/busio.py", line 89, in writeto_then_readfrom
    in_start=in_start, in_end=in_end, stop=stop)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 61, in writeto_then_readfrom
    readin = self._i2c_bus.read_i2c_block_data(address, buffer_out[out_start:out_end], in_end-in_start)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/smbus.py", line 207, in read_i2c_block_data
    reg = c_uint8(cmd)
TypeError: an integer is required (got type bytearray)`

caternuson, 2020-01-09:
How repeatable is that error? I can't recreate this. I brought up a Pi Zero W with the 2019-09-26 Raspbian Buster Lite OS image, went through the Blinka install, and tested your same code and it ran fine:
```python
(blinka) pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux
(blinka) pi@raspberrypi:~$ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> import busio
>>> import adafruit_ads1x15.ads1115 as ADS
>>> from adafruit_ads1x15.analog_in import AnalogIn
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> ads = ADS.ADS1115(i2c)
>>> chan = AnalogIn(ads, ADS.P0)
>>> print(chan.value, chan.voltage)
4664 0.5857678762169256
>>>
```

Grandpa-G, 2020-01-09:
I just ran the exact script you have above and got the error I described. I also did the same uname -a command and got the same results.

caternuson, 2020-01-09:
Is there anything unique about your setup? The version of Python you are running, etc.? Or is this essentially a fresh install of the Raspbian OS?

Did you follow this guide to install Blinka:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/overview
and then this guide for the ADS1115:
https://learn.adafruit.com/adafruit-4-channel-adc-breakouts/python-circuitpython

Grandpa-G, 2020-01-09:
It is the latest update of Raspbian OS so it is also latest Python3. I use sudo pip3 to install per the directions but I don't use environment.

It is not exactly a virgin OS. I have added some other libraries, but I really can't tell  you which as I never really kept track. If I get time I will try a new OS. Unfortunately, the project moved on and my motivation has diminished, but I will try to help out.

Grandpa-G, 2020-01-09:
I didn't mean to close this, sorry.

caternuson, 2020-01-09:
There are a few other open issues that are hitting this same thing. So there must be some underlying issue here. It's just not easy to repeat by starting with a fresh OS and working forward from there. So if we can figure out what's unique about your setup and how you got there, that would help.

Go ahead and keep this open. When you have more info, please share. Thanks.

Grandpa-G, 2020-01-09:
I know for sure I installed the pyqt5 library.

I am starting from fresh img now. I will add stuff and see if I can break it.

Grandpa-G, 2020-01-09:
I rebuilt the img from a dowload 9-25-19. I then went through every step in the install Blinka document and then install the ADS1115 per the guide. I didn't add anything else.

I am having 1 problem that my device isn't showing up on i2cdetect, but I think it is a loose wire.

I am using the same small program you have.

When I get i2cdetect to work, I will give you the results.

Grandpa-G, 2020-01-09:
This setup failed as before. No extra software installed. Using ssh to connect.

If I can get my PiZero connected, I will try it.

caternuson, 2020-01-09:
OK, I'll try again and this time won't use a venv. Just to make sure - were you using Raspbian Lite or Full?

Grandpa-G, 2020-01-09:
I just put it on my piZero and it doesn't crash.

I am going to put it back on my Pi3 now.

caternuson, 2020-01-09:
Interesting. I should have asked what model Pi also. My original test was a on Pi Zero W and it sounds like you've repeated success there. So maybe something Pi model specific. Was the Pi3 what you were using originally in your first post above?

Grandpa-G, 2020-01-09:
I am using Raspbian full. It is a late model Pi 3+ (or whatever it is called, not a 4)

Grandpa-G, 2020-01-09:
Now it's time to scratch one's head. I put it back on the original Pi 3 I had errors. It runs now. The only thing I did was re-soldered the power pin on the ADS1115. Maybe that was the problem all along.

I will next go back to the original OS setup and see if it works.

Grandpa-G, 2020-01-09:
Going back to the OS where the problem started, I still have the error.

Exact same hardware as the step before that worked. I am not sure what I can do to help now. I might try to rebuild some of the parts I installed for other parts of my project.

Sorry I can't be of more help.

caternuson, 2020-01-09:
Thanks for the testing. This seems like more of software issue than a hardware issue. Somehow that one OS image has gotten in a weird state where there is some kind of software conflict going on. Trying to figure out what and how it happened can be tricky. But by starting with a fresh OS, you know everything works in general.

The person over in [this other issue thread](https://github.com/adafruit/Adafruit_CircuitPython_VCNL4040/issues/7#issuecomment-572712119) was also able to get things to work by simply starting with a fresh OS image.

Let's close this issue for now. That doesn't lock it. You can still come back and add new comments and information. If you run into this again and can remember what the last thing you did was, like "installed so and so package", then that would help. We can reopen the issue then and see if we can figure out more.

Thanks for going through the troubleshooting above!

closed 2020-01-09 by caternuson

caternuson, 2020-01-10:
@Grandpa-G Here's something you can try. It worked in the other thread:
https://github.com/adafruit/Adafruit_CircuitPython_VCNL4040/issues/7#issuecomment-573195267

Grandpa-G, 2020-01-10:
This was a good solution. For me, I had to add sudo and change to pip3, but the program now runs. Here is the output so you can see only a few libraries were updated. This should be the clue to what really needs to be done.

pi@raspberrypi:~ $ sudo pip3 freeze --local | grep -v '^\-e' | cut -d = -f 1  | xargs -n1 pip3 install -U
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: Adafruit-ADS1x15 in ./.local/lib/python3.7/site-packages/Adafruit_ADS1x15-1.0.2-py3.7.egg (1.0.2)
Requirement already satisfied, skipping upgrade: Adafruit-GPIO>=0.6.5 in /usr/local/lib/python3.7/dist-packages (from Adafruit-ADS1x15) (1.0.3)
Requirement already satisfied, skipping upgrade: spidev in /usr/local/lib/python3.7/dist-packages (from Adafruit-GPIO>=0.6.5->Adafruit-ADS1x15) (3.4)
Requirement already satisfied, skipping upgrade: adafruit-pureio in /usr/local/lib/python3.7/dist-packages (from Adafruit-GPIO>=0.6.5->Adafruit-ADS1x15) (0.2.3)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting Adafruit-Blinka
  Downloading https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-3.3.10-py3-none-any.whl (86kB)
    100% |████████████████████████████████| 92kB 253kB/s
Requirement already satisfied, skipping upgrade: Adafruit-PureIO in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (0.2.3)
Requirement already satisfied, skipping upgrade: sysv-ipc; platform_system != "Windows" and platform_machine != "mips" in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (1.0.1)
Requirement already satisfied, skipping upgrade: Adafruit-PlatformDetect in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (1.3.8)
Requirement already satisfied, skipping upgrade: rpi-ws281x>=4.0.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (4.2.2)
Requirement already satisfied, skipping upgrade: RPi.GPIO in /usr/lib/python3/dist-packages (from Adafruit-Blinka) (0.7.0)
Requirement already satisfied, skipping upgrade: spidev>=3.4; sys_platform == "linux" and platform_machine != "mips" in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (3.4)
Requirement already satisfied, skipping upgrade: pyftdi>=0.40.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka) (0.42.2)
Requirement already satisfied, skipping upgrade: pyusb>=1.0.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka) (1.0.2)
Requirement already satisfied, skipping upgrade: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka) (3.4)
Installing collected packages: Adafruit-Blinka
Successfully installed Adafruit-Blinka-3.3.10
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: adafruit-circuitpython-ads1x15 in /usr/local/lib/python3.7/dist-packages (2.1.1)
Requirement already satisfied, skipping upgrade: Adafruit-Blinka in ./.local/lib/python3.7/site-packages (from adafruit-circuitpython-ads1x15) (3.3.10)
Requirement already satisfied, skipping upgrade: adafruit-circuitpython-busdevice in /usr/local/lib/python3.7/dist-packages (from adafruit-circuitpython-ads1x15) (4.1.1)
Requirement already satisfied, skipping upgrade: rpi-ws281x>=4.0.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-ads1x15) (4.2.2)
Requirement already satisfied, skipping upgrade: RPi.GPIO in /usr/lib/python3/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-ads1x15) (0.7.0)
Requirement already satisfied, skipping upgrade: Adafruit-PureIO in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-ads1x15) (0.2.3)
Requirement already satisfied, skipping upgrade: sysv-ipc; platform_system != "Windows" and platform_machine != "mips" in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-ads1x15) (1.0.1)
Requirement already satisfied, skipping upgrade: spidev>=3.4; sys_platform == "linux" and platform_machine != "mips" in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-ads1x15) (3.4)
Requirement already satisfied, skipping upgrade: Adafruit-PlatformDetect in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-ads1x15) (1.3.8)
Requirement already satisfied, skipping upgrade: pyftdi>=0.40.0 in /usr/local/lib/python3.7/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-ads1x15) (0.42.2)
Requirement already satisfied, skipping upgrade: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-ads1x15) (3.4)
Requirement already satisfied, skipping upgrade: pyusb>=1.0.0 in /usr/local/lib/python3.7/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-ads1x15) (1.0.2)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: adafruit-circuitpython-busdevice in /usr/local/lib/python3.7/dist-packages (4.1.1)
Requirement already satisfied, skipping upgrade: Adafruit-Blinka in ./.local/lib/python3.7/site-packages (from adafruit-circuitpython-busdevice) (3.3.10)
Requirement already satisfied, skipping upgrade: RPi.GPIO in /usr/lib/python3/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-busdevice) (0.7.0)
[... cut ...]
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: sysv-ipc in /usr/local/lib/python3.7/dist-packages (1.0.1)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: websockets in /usr/local/lib/python3.7/dist-packages (8.1)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: xmodem in /usr/local/lib/python3.7/dist-packages (0.4.5)