adafruit/Adafruit_CircuitPython_ADS1x15 #50: ADS1115 works on my PI4, not PI3B

smclaugh5, 2020-01-14:
mport board
import busio
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.analog_in import AnalogIn
from time import sleep

i2c = busio.I2C(board.SCL, board.SDA)
ads = ADS.ADS1115(i2c)
#6.144 4.096 2.048 1.024 0.512 0.256 Volts:
ads.gain = 2/3   #or 1,2,4,8,16 (don't know if 2/3 is syntax?)
chan = AnalogIn(ads, ADS.P0)
print(chan.value, chan.voltage)

i = 0
while i < 250:
    print(chan.value, chan.voltage)
    i += 1
    sleep (0.15)

On 3B gives the following errors:

Traceback (most recent call last):
  File "/home/pi/Documents/python_programs/16bita2d_1.py", line 12, in <module>
    print(chan.value, chan.voltage)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ads1x15/analog_in.py", line 74, in value
    is_differential=self.is_differential) << (16 - self._ads.bits)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ads1x15/ads1x15.py", line 137, in read
    return self._read(pin)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ads1x15/ads1x15.py", line 166, in _read
    while not self._conversion_complete():
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ads1x15/ads1x15.py", line 176, in _conversion_complete
    return self._read_register(_ADS1X15_POINTER_CONFIG) & 0x8000
  File "/usr/local/lib/python3.7/dist-packages/adafruit_ads1x15/ads1x15.py", line 202, in _read_register
    i2c.write_then_readinto(bytearray([reg]), self.buf, in_end=2)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 136, in write_then_readinto
    in_start=in_start, in_end=in_end)
  File "/usr/local/lib/python3.7/dist-packages/busio.py", line 91, in writeto_then_readfrom
    in_start=in_start, in_end=in_end, stop=stop)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 61, in writeto_then_readfrom
    readin = self._i2c_bus.read_i2c_block_data(address, buffer_out[out_start:out_end], in_end-in_start)
  File "/usr/local/lib/python3.7/dist-packages/Adafruit_PureIO/smbus.py", line 207, in read_i2c_block_data
    reg = c_uint8(cmd)
TypeError: an integer is required (got type bytearray)

caternuson, 2020-01-14:
See here for possible fix:
https://github.com/adafruit/Adafruit_CircuitPython_VCNL4040/issues/7#issuecomment-573195267
and here for more discussion:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/issues/48

We are not sure what is causing this, but seems to result from getting your environment into a certain state. If you can provide more info on how the Pi 3B got there, that would help.

closed 2020-01-14 by ladyada

smclaugh5, 2020-01-14:
OK I am following the other threads about python update and fresh OS install.

I have a system with a multiplexer and 6 DAC's, plus the A-2-D.  Everything else works fine.  Added the A-2-D (1115) to another I2C plexer channel.  Error.  Then bypassed the multiplexer.  Same error.  Then disconnected 5V to 1115 / 1015.  Same error.  But powered up, if I swap the I2C leads, I get a nothing on the bus error.  So seems that the part wants to talk to PI, but software related.

I am going to make an image backup.  Then try:

pip freeze --local | grep -v '^\-e' | cut -d = -f 1  | xargs -n1 pip install -U

smclaugh5, 2020-01-14:
sudo pip3 freeze --local | grep -v '^-e' | cut -d = -f 1 | xargs -n1 pip3 install -U

Fixed problem.  Thank you so much!