adafruit/Adafruit_CircuitPython_ADS1x15 #51: v = (chan.voltage) dies as a service

smclaugh5, 2020-01-28:
I can get the voltage no problem with:

            chan = AnalogIn(ads, ADS.P0)   #channel 1 of 4 on 16 bit A2D ADS1115
            v = (chan.voltage) 

but I have the python program set up as a linux service so I can get my data acquisition program
to execute automatically on boot.  The (chan.voltage) kills the script. The program runs fine from the command line or the Thonny environment.  Any suggestions?

caternuson, 2020-01-28:
In general, you do not need the parans. I doubt that is the issue, but you can do:
```python
v = chan.voltage
```
If your intent is to create a single member tuple, you would need to do:
```python
v = (chan.voltage, )
```

Do you have a stack trace or any other kind of error message for what happens when the script fails to run?

smclaugh5, 2020-01-29:
test_adc.service - test the adc as a service
   Loaded: loaded (/lib/systemd/system/test_adc.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2020-01-29 14:21:47 CST; 6s ago
  Process: 1545 ExecStart=/usr/bin/python3 /home/pi/Documents/python_programs/adc_test.py (code=exited, status=1/FAILURE)
 Main PID: 1545 (code=exited, status=1/FAILURE)
*******************************************************************************
Here is the code:
#!/usr/bin/python3
from time import sleep
import smbus
import board
import busio
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.analog_in import AnalogIn
import Adafruit_MCP4725

class multiplex:
    
    def __init__(self, bus):
        self.bus = smbus.SMBus(bus)

    def channel(self, address,channel):  # values 0-7 indictae the channel, anything else (eg -1) turns off all channels
        
        if   (channel==0): action = 0x01
        elif (channel==1): action = 0x02
        elif (channel==2): action = 0x04
        elif (channel==3): action = 0x08
        elif (channel==4): action = 0x10
        elif (channel==5): action = 0x20
        elif (channel==6): action = 0x40
        elif (channel==7): action = 0x80
        else : action = 0x00

        self.bus.write_byte_data(address,0x04,action)  #0x04 is the register for switching channels 

bus=1       
address=0x70  
plexer = multiplex(bus)

plexer.channel(address,6)  #multiplexer channel for 16 bit a2d
i2c = busio.I2C(board.SCL, board.SDA)
ads = ADS.ADS1115(i2c)
ads.gain = 2/3   

# Create a DAC instance.
dac = Adafruit_MCP4725.MCP4725()

plexer.channel(address,6)  #multiplexer channel for 16 bit a2d
chan = AnalogIn(ads, ADS.P0)   #channel 1 of 4 on 16 bit A2D ADS1115
v = (chan.voltage)  
v = chan.voltage  #works either way 
psi1115 = (chan.voltage / 5)*300 -15 #(correction for absolute)
print("Voltage = ", v, "Pressure = ", psi1115)

while True:
    print("Voltage = ", v, "Pressure = ", psi1115)
    sleep(1)

***********************************************************************************
here is the service:

[Unit]
Description=test the adc as a service

[Service]
Environment=DISPLAY=:0
Environment=XAUTHORITY=/home/pi/.Xauthority
ExecStart=/usr/bin/python3 /home/pi/Documents/python_programs/adc_test.py
Restart=always
RestartSec=10s
KillMode=process
TimeoutSec=infinity

[Install]
WantedBy=graphical.target

***********************************************************************************************
I can see the I2C bus activity on an oscilloscope every 10 second.
***********************************************************************************************
output from py program in editing environment:
Voltage =  0.5242659993285927 Pressure =  15.375926999725333
Voltage =  0.5242659993285927 Pressure =  15.375926999725333
Voltage =  0.5242659993285927 Pressure =  15.375926999725333
Voltage =  0.5242659993285927 Pressure =  15.375926999725333
Voltage =  0.5242659993285927 Pressure =  15.375926999725333
Voltage =  0.5242659993285927 Pressure =  15.375926999725333
Voltage =  0.5242659993285927 Pressure =  15.375926999725333

smclaugh5, 2020-01-29:
Sorry, some of that uploaded kinda weird....

caternuson, 2020-01-29:
Based on that stack trace, what makes you think it's related to `chan.voltage`? Looks like all we know is it exited with a non-zero status:
```
Process: 1545 ExecStart=/usr/bin/python3 /home/pi/Documents/python_programs/adc_test.py (code=exited, status=1/FAILURE)
```
It could be something like an import error due to that process running in a different environment than when running via command line or Thonny.

smclaugh5, 2020-01-29:
Comment out chan.voltage and the program runs.

caternuson, 2020-01-29:
Where do you comment that out? It's used in several places. Can you recreate this with a more simple example that just uses the ADS1115?

You can also edit your post above with the code to clean up the code syntax.

smclaugh5, 2020-01-30:
I am working on a simple set up.  It is interesting to note that it won't survive a try / except either.

Standby for more information;  To answer your prior question:

v = chan.voltage

The above commented out allows the code to run as a sub process or service.

smclaugh5, 2020-01-31:
OK, looks like the problem is related to using the old TCA9548A multiplexer example python code and modules with the newer Circuit_Python modules for the ADC.  I am going to close this (unresolved) thread and try to get some help with setting up the multiplexer and the DAC's and ADC's.  

Still confused why the code works in then environment and not as a service or a sub-process.

smclaugh5, 2020-01-31:
BTW, the ADC works fine as a stand alone connected to the PI i2c.  Just not through the TCA9548A.

caternuson, 2020-01-31:
Tested and seems to work. If you continue to have issues, please post in the [forums](https://forums.adafruit.com/) with photos of your setup.

[image]

```python
(blinka) pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> import adafruit_tca9548a
>>> import adafruit_ads1x15.ads1115 as ADS
>>> from adafruit_ads1x15.analog_in import AnalogIn
>>> tca = adafruit_tca9548a.TCA9548A(board.I2C())
>>> adc = ADS.ADS1115(tca[0])
>>> chan = AnalogIn(adc, ADS.P0)
>>> chan.voltage
0.888277108066042
>>> chan.voltage
1.5059209570604573
>>> chan.voltage
2.4190738242744225
>>> chan.voltage
1.1637855159154027
>>> 
```

closed 2020-01-31 by caternuson