adafruit/Adafruit_CircuitPython_ADS1x15 #58: Continuous Sampling from more than one channel ADS1015`

rmiller4, 2020-08-05:
Hello, 
I've been using the ads1x15_fast_read.py code and I am trying to read
from two channels continuously in single-ended mode.
I simply created a second channel (chan1 = AnalogIn(ads, ADS.P1)
in addition to the first, along with another data array (dataB). Then I read them
one after the other using 
data[i] = chan0.value
dataB[i] = chan1.value

When I do this, the values for both arrays seems to be nonsensical. 
Is there a way to read in more than one channel in conintuous/fast mode?
Or am I missing something obvious. 
Thanks for any help.

rmiller4, 2020-08-05:
I just read the previous threads (esp. issue #57 ). It seems I was late to the party...does anyone know if there's a c library that would allow for fast reads (single-ended) from multiple channels in continuous mode?

dherrada, 2021-01-22:
@rmiller4 Sorry for the late response. I'd try [this library](https://github.com/adafruit/Adafruit_ADS1X15.git)

jposada202020, 2021-04-23:
@rmiller4 did you verify the library mentioned above? let me know thanks :)

jposada202020, 2021-05-07:
closing this for now. thanks