adafruit/Adafruit_CircuitPython_ADS1x15 #60: ad1115 totally weird values when overlay ad1115 is not loaded

marcelser, 2020-10-11:
Hi,

I'm trying to use the ADS1115 driver from Circuitpython to replace the outdate old 1.0.2 one. However it seems I need to load the also outdated AD1115 overlay to make the return values make sense. Otherwise the AD1115 outputs just nonsense the values are jumping around from 20000+ to -116 in all channels.

When I add the ADS1115 overlay everthing looks perfectly fine. So it seems the overlay does some tricks (probably initialization wise) which CircuitPython is missing. How can I get rid of the overlay, do we need a better initialization function?

Cause when using dmesg output shows that ad1115 overlay is also outdated as it uses a deprecation driver add method.

Best,
Marc

caternuson, 2021-06-30:
Are you referring to device tree overlays? What platform are you using the ADS on? Sounds like maybe this a Raspberry Pi?

In general, use of the ASD1115 does not require any special drivers or dto's. You only need to enable I2C. Everything is done over I2c via this library.

caternuson, 2022-05-17:
Closing due to lack of response.

closed 2022-05-17 by caternuson