adafruit/Adafruit_CircuitPython_ADS1x15 #63: Raspberry Pi 4 Model B Rev 1.4

TobBvB09, 2020-12-16:
Hello,

I'm trying to use the ads1x15 library with my Raspberry Pi 4 Model B Rev 1.4, but when I run any of the examples I get this ERROR:
```
Traceback (most recent call last):
  File "ads1x15_simpletest.py", line 2, in <module>
    import board
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 166, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None
```

Further info of my system:
```
Package                          Version
-------------------------------- ---------
Adafruit-ADS1x15                 1.0.2
Adafruit-Blinka                  5.9.1
adafruit-circuitpython-ads1x15   2.2.4
adafruit-circuitpython-busdevice 5.0.1
Adafruit-GPIO                    1.0.3
Adafruit-PlatformDetect          2.23.0
Adafruit-PureIO                  1.1.8
```

```
cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : b03114
Serial          : 100000006a43b54f
Model           : Raspberry Pi 4 Model B Rev 1.4
```
On the Raspberry Pi 4 Model B Rev 1.1 it worked and I think the only thing that changed is the model revision.
Do you have any ideas?

caternuson, 2020-12-16:
That rev code should be supported:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/168ff6b66548e8873d5e05ecbe4f24d2e8b22124/adafruit_platformdetect/constants/boards.py#L395

When did you install Blinka? Did you pass the post install check?
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#blinka-test-3030038-17

TobBvB09, 2020-12-17:
I have reinstalled my raspberry pi again and updated everything. Now the blinkatest and everything else works. 
Probably I just had an old version.
Thanks!

closed 2020-12-17 by TobBvB09