adafruit/Adafruit_CircuitPython_ADS1x15 #64: need support for user-defined I2C,  channels > 1

jrbrearley, 2021-01-02:
I was able to add i2c-4 bus in the /boot/config.txt file with:
    dtoverlay=i2c-gpio,bus=4,i2c_gpio_delay_us=1,i2c_gpio_sda=17,i2c_gpio_scl=27
 
see I2C tutorial: https://www.instructables.com/Raspberry-PI-Multiple-I2c-Devices

sudo i2cdetect -y 4 correctly show the attached ADS1115 on I2C bus 4 at address 0x.48

However python adafruit_ads1x15 does not like i2c bus 4

import board, busio, time, traceback
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.analog_in import AnalogIn
i2c_1 = busio.I2C(board.SCL, board.SDA) # works OK
# i2c_4 = busio.I2C(scl=13, sda=11) # gets error
i2c_4 = busio.I2C(board.D13, board.D11) # gets error

Traceback (most recent call last):
File "ads1115_read_ch.py", line 8, in <module>
i2c_4 = busio.I2C(board.D13, board.D11)
File "/home/pi/.local/lib/python3.7/site-packages/busio.py", line 31, in __init__
self.init(scl, sda, frequency)
File "/home/pi/.local/lib/python3.7/site-packages/busio.py", line 72, in init
(scl, sda), i2cPorts
ValueError: No Hardware I2C on (scl,sda)=(13, 11)
Valid I2C ports: ((3, 3, 2), (1, 3, 2), (0, 1, 0))

To convince myself that I2c bus 4 is real, I wrote a sample script below directly calling smbus routines directly.

import smbus, time, traceback

# Get I2C bus 1 & 4
bus1 = smbus.SMBus(1)
# bus3 = smbus.SMBus(3) # i2c bus 3 is not defined on my system, gets error, as expected
bus4 = smbus.SMBus(4)
print("found bus1: ", bus1, " bus4: ", bus4,"\n")

while True:
   try:
      # Read data from two i2c bus
      b11 = bus1.read_i2c_block_data(0x48, 0x48, 8) # (bus_address, register, length)
      b14 = bus1.read_i2c_block_data(0x4b, 0x48, 8)
      b41 = bus4.read_i2c_block_data(0x48, 0x48, 8)
      print("b11: ", b11, "\nb14: ", b14, "\nb41: ",b41)
      time.sleep(0.5)

   except KeyboardInterrupt:
      print("\n\nexiting.")
      break

   except:
      print("Start exception handler")
      traceback.print_exc()
      print("End exception handler")

python3 read_i2c_bus.py
found bus1:  <SMBus object at 0x7669cf80>  bus4:  <SMBus object at 0x766e8d28>

b11:  [20, 47, 255, 255, 255, 255, 255, 255]
b14:  [17, 137, 255, 255, 255, 255, 255, 255]
b41:  [0, 0, 255, 255, 255, 255, 255, 255]
b11:  [20, 47, 255, 255, 255, 255, 255, 255]
b14:  [17, 137, 255, 255, 255, 255, 255, 255]

Admittedly I am reading the wrong register for the actual voltage, but bus4 is an I2C bus.

It would be really nice to have adafruit support more than one I2C bus.

ladyada, 2021-01-02:
1) please format your issues with code blocks so they are readable

2) you can use https://github.com/adafruit/Adafruit_Python_Extended_Bus for making more buses

closed 2021-01-02 by ladyada

ladyada, 2021-01-02:
(if you have any issue with the above solution plz post an issue in the extended bus repo! :)