adafruit/Adafruit_CircuitPython_ADS1x15 #66: adafruit_ads1x15.ads1115 not comaptible with globals()[pkg_trunc] = importlib.import_module(pkg_trunc)

jrbrearley, 2021-01-13:
I have sorted out my object syntax issues with adafruit_ads1x15.ads1115. The first attached script ada_dbg1.py.txt which uses simple hardcoded import statements, the creation of ads1115 objects work fine. The second attached script ada_dbg2.py.txt uses a loop to import a variable list of packages, which allows for more graceful error handling & messaging.

`globals()[pkg_trunc] = importlib.import_module(pkg_trunc) 
`

The high level observation is that board, busio & adafruit_extended_bus objects continue to be correctly created after the import loop is done. Its just adafruit_ads1x15.ads1115 that does not like this import method.

The same line of code creating ads10 gets error:
Traceback (most recent call last):
File "ada_dbg2.py", line 33, in <module>
ads10 = adafruit_ads1x15.ads1115.ADS1115(i2c_1, address=0x48)
AttributeError: module 'adafruit_ads1x15' has no attribute 'ads1115'

When I look at the global variables, from ads_dbg1.py, I see:
'adafruit_ads1x15': <module 'adafruit_ads1x15' from '/home/pi/.local/lib/python3.7/site-packages/adafruit_ads1x15/__init__.py'>,

When I look at the global variables, from ads_dbg2.py, I see:
'adafruit_ads1x15.ads1115': <module 'adafruit_ads1x15.ads1115' from '/home/pi/.local/lib/python3.7/site-packages/adafruit_ads1x15/ads1115.py'>,

So I tried doing the import without the .ads1115 appended, & got:
'adafruit_ads1x15': <module 'adafruit_ads1x15' from '/home/pi/.local/lib/python3.7/site-packages/adafruit_ads1x15/__init__.py'>,

But I still get an error.
Traceback (most recent call last):
File "ada_dbg2.py", line 33, in <module>
ads10 = adafruit_ads1x15.ads1115.ADS1115(i2c_1, address=0x48)
AttributeError: module 'adafruit_ads1x15' has no attribute 'ads1115'

I went through all 4 permutations of with/without the .ads1115 in the  importlib.import_module statement, no luck.

Comments?

[ada_dbg1.py.txt](https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/files/5809553/ada_dbg1.py.txt)
[ada_dbg2.py.txt](https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/files/5809554/ada_dbg2.py.txt)
[print_data.py.txt](https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/files/5809555/print_data.py.txt)

ladyada, 2021-01-13:
we arent clear that this has anything to do with the library structure or is just a Pythonism....
as the joke goes "if it hurts when you do that, stop doing that" ? use imports the way Python intended em to be used?

jrbrearley, 2021-01-14:
Yes, I can work around this in the short/long term.

BTW: I tried the import_module( ... , package=<pkg>) option, nothing useful happened.

To get a wider viewing of the issue, I posted this article at: https://bugs.python.org/issue42928

dhalbert, 2021-01-14:
`adafruit_ads1x15.ads1115` is the only module you're importing in this non-standard way that is of the form `a.b` (dot-separated). I don't know why `importlib` doesn't like that, but I don't think it has to do with our library. Try a simpler example of some dummy code where you are importing something that has `a.b`. But this is a Python question, not a library issue, as far as I can tell. Our library is not doing anything unusual.

closed 2021-01-14 by dhalbert

jrbrearley, 2021-01-17:
I tried loading adafruit_ads1x15.analog_in via importlib.import_module and it works fine, see  attached power_monitor_rpi.py.txt below. 

When you run the script, the globals data shows:

```
'adafruit_extended_bus': <module 'adafruit_extended_bus' from '/home/pi/.local/lib/python3.7/site-packages/adafruit_extended_bus.py'>, 

'adafruit_ads1x15.analog_in': <module 'adafruit_ads1x15.analog_in' from '/home/pi/.local/lib/python3.7/site-packages/adafruit_ads1x15/analog_in.py'>, 

'adafruit_ads1x15': <module 'adafruit_ads1x15' from '/home/pi/.local/lib/python3.7/site-packages/adafruit_ads1x15/__init__.py'>}

```
The first two packages point to the lowest level .py file with code in them, but the adafruit_ads1x15 points to an (empty) --init--.py file. Maybe this is a clue?

[power_monitor_rpi.py.txt](https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/files/5826453/power_monitor_rpi.py.txt)