adafruit/Adafruit_CircuitPython_ADS1x15 #69: how to read two ADS1115 on I2C bus?

dretful, 2021-02-24:
With two ADS1115's connected directly to the i2c bus and using the option to instantiate them with different addresses, I cant read each of them (having followed the example syntax).

    ads1 = ADS.ADS1115(i2c, address=0x48)
    chan1 = AnalogIn(ads1, ADS.P0)
    ads2 = ADS.ADS1115(i2c, address=0x49)
    chan2 = AnalogIn(ads2, ADS.P0)
    print("ads1 {:>5}\t{:>5.3f}".format(chan1.value, chan1.voltage))
    print("ads2 {:>5}\t{:>5.3f}".format(chan2.value, chan2.voltage))

output

ads1 4673 0.584 
Traceback (most recent call last): File "<stdin>",
line 173, in <module> File adafruit_ads1x15/analog_in.py", 
line 47, in value File "adafruit_ads1x15/ads1x15.py", 
line 127, in read File "adafruit_ads1x15/ads1x15.py", 
line 162, in _read File "adafruit_ads1x15/ads1x15.py", 
line 198, in _write_register File "adafruit_ads1x15/ads1x15.py", 
line 198, in _write_register File "adafruit_bus_device/i2c_device.py", 
line 84, in write OSError: [Errno 19] Unsupported operation

What's wrong here?

caternuson, 2021-02-24:
Did you use the ADDR pin to set the alt address for the second ADS? Try an I2C scan and make sure both are being seen.

ladyada, 2021-02-27:
hihi -this should be posted in the tech support forums

closed 2021-02-27 by ladyada

dretful, 2021-02-28:
Sorry of wrong place, but just wanted to say, yes thank you following the advice `Did you use the ADDR pin to set the alt address for the second ADS? Try an I2C scan and make sure both are being seen.` and that solved it for me.