adafruit/Adafruit_CircuitPython_ADS1x15 #70: ads1115 gets stuck in the following situation

fabioHernandezRu, 2021-06-29:
I am testing the library in rpi3 b, but I find that the program remains in an infinite loop if it is not able to read the ADC because i2c fails .. how could this be avoided?

ladyada, 2021-06-29:
i2c must be valid, it should not be disconnected

caternuson, 2021-06-30:
Is the failure something other than disconnection? What exactly is the nature of the failure?

fabioHernandezRu, 2021-06-30:
I think is the i2c bus because when I write I2C detect I get:
` 0 1 2 3 4 5 6 7 8 9 a b c d e f 00: 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70: 70 71 72 73 74 75 76 77`

the problem is that this failure cause and infinity loop inside the library.

caternuson, 2021-06-30:
Yep, need to figure out what is causing that. Looks like a hardware issue. And in general, in that state, there's not much this library can do to work around it.

closed 2021-06-30 by fabioHernandezRu