adafruit/Adafruit_CircuitPython_ADS1x15 #80: Sampling rate changes in single mode when changing datarate but not in continuous mode

konstntokas, 2022-05-05:
Hi, 

the code below shows a little example to test the influence of the datarate. When I am in single mode and adjust the sampling rate, I get the following result.

- 8: ~50 SPS
- 16: ~90 SPS
- 64: ~160 SPS
- 128: ~128 SPS

When I use  continuous mode and adjust the datarate, nothing changes at all. 
Can somebody explain this? According to the datasheet of ADS1115 I expected that the datarate does not have an influence in single mode but in continuous. And why does the datarate does not fit to the actual SPS? Many Thanks!

`import board
import busio
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.analog_in import AnalogIn
from adafruit_ads1x15.ads1x15 import Mode
import time
import datetime

i2c = busio.I2C(board.SCL, board.SDA)

# init ads instance
ads = ADS.ADS1115(i2c)
channel = AnalogIn(ads, ADS.P0, ADS.P1)

# ADC Configuration
ads.mode = Mode.SINGLE
ads.data_rate = 8

ts_old = datetime.datetime.now()
samples = 1000
for i in range(samples):
    ts = datetime.datetime.now()
    print(str(channel.voltage))
    print(ts - ts_old)
    ts_old = ts`

caternuson, 2022-05-05:
This is related to how the library handles SINGLE mode vs. CONTINUOUS mode. See the discussion here:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/issues/27

What is your specific goal in terms of reading the ADS1115? Do you require super-fast-reading? Or are you just curious about the noted behavior?

konstntokas, 2022-05-10:
Thanks fro redirection to #27 . This explains more.  In the beginning I have the understanding that if the data rate is low, internal averaging is performed. But this is wrong as far as I can see. Now I am just curious. 
So in CONTINUOUS mode the values are read as quick as possible? However I still do not understand why in SINGLE mode the data rate has an influence, because it is a single shot/read on demand.

caternuson, 2022-05-10:
Those kinds of details are best determined from the datasheet.
https://learn.adafruit.com/adafruit-4-channel-adc-breakouts/downloads

closed 2022-05-10 by caternuson