adafruit/Adafruit_CircuitPython_ADS1x15 #81: SINGLE and CONTINUOUS mode have different voltage value

Yingliangzhe, 2022-05-17:
Hello everyone, 
I am a little bit confused by the SINGLE and CONTINUOUS mode of ads1115.
I use the following code to test.

```python
import time
import board
import busio
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.ads1x15 import Mode
from adafruit_ads1x15.analog_in import AnalogIn

i2c = busio.I2C(board.SCL, board.SDA)
ads = ADS.ADS1115(i2c)
ads.mode = Mode.CONTINUOUS
#ads.mode = Mode.SINGLE
ads.gain = 1
voltage_divider = 0.4

chan1 = AnalogIn(ads, ADS.P0)
chan2 = AnalogIn(ads, ADS.P1)
chan3 = AnalogIn(ads, ADS.P2)
chan4 = AnalogIn(ads, ADS.P3)

while True:
  #chan1 = AnalogIn(ads, ADS.P0)
  #print(ADS.P0)
  print(f'0. channel value is: {chan1.value}, channel voltage is: {chan1.voltage/voltage_divider}')

  #chan2 = AnalogIn(ads, ADS.P1)
  print(f'1. channel value is: {chan2.value}, channel voltage is: {chan2.voltage/voltage_divider}')

  #chan3 = AnalogIn(ads, ADS.P2)
  print(f'2. channel value is: {chan3.value}, channel voltage is: {chan3.voltage/voltage_divider}')

  #chan4 = AnalogIn(ads, ADS.P3)
  print(f'3. channel value is: {chan4.value}, channel voltage is: {chan4.voltage/voltage_divider}')

  print('\n')

  time.sleep(1)
```

If I use the CONTINUOUS mode, the output is like this:

```
0. channel value is: 8638, channel voltage is: 5.321099887081515
1. channel value is: 8679, channel voltage is: 6.601138950773644
2. channel value is: 8908, channel voltage is: 7.881178014465774
3. channel value is: 8742, channel voltage is: 9.161217078157902

0. channel value is: 8638, channel voltage is: 5.321099887081515
1. channel value is: 8680, channel voltage is: 6.601138950773644
2. channel value is: 8907, channel voltage is: 7.881178014465774
3. channel value is: 8742, channel voltage is: 9.161217078157902

0. channel value is: 8638, channel voltage is: 5.321099887081515
1. channel value is: 8679, channel voltage is: 6.601138950773644
2. channel value is: 8908, channel voltage is: 7.881178014465774
3. channel value is: 8741, channel voltage is: 9.161217078157902

0. channel value is: 8634, channel voltage is: 5.321099887081515
1. channel value is: 8679, channel voltage is: 6.601138950773644
2. channel value is: 8908, channel voltage is: 7.881178014465774
3. channel value is: 8742, channel voltage is: 9.161217078157902
```

If I switch the mode the SINGLE, it gives me the following:

```
0. channel value is: 8741, channel voltage is: 2.6994573809015163
1. channel value is: 8742, channel voltage is: 2.712270271919919
2. channel value is: 8679, channel voltage is: 2.7838349559007534
3. channel value is: 8742, channel voltage is: 2.7319583727530747

0. channel value is: 8679, channel voltage is: 2.6994573809015163
1. channel value is: 8679, channel voltage is: 2.710707724234748
2. channel value is: 8908, channel voltage is: 2.782897427289651
3. channel value is: 8742, channel voltage is: 2.7319583727530747

0. channel value is: 8638, channel voltage is: 2.6994573809015163
1. channel value is: 8679, channel voltage is: 2.712270271919919
2. channel value is: 8908, channel voltage is: 2.7838349559007534
3. channel value is: 8741, channel voltage is: 2.7316458632160403

0. channel value is: 8741, channel voltage is: 2.698519852290414
1. channel value is: 8679, channel voltage is: 2.712270271919919
2. channel value is: 8908, channel voltage is: 2.7838349559007534
3. channel value is: 8739, channel voltage is: 2.7316458632160403

```

The measurement values from SINGLE mode are actually right. 
Does someone know, what is the problem? 

Thank you!

caternuson, 2022-05-17:
The raw count values generally agree between the two outputs. Are you able to vary the input voltage? If so, try that and see if the raw values track as expected - that is, see if you get the same readings in terms of ADC counts with both SINGLE and CONTINUOUS with some new input voltage.

Yingliangzhe, 2022-05-17:
@caternuson 
I actually expect some noise from the each measurement. But it seems the same. 
So I thought, if I should use CONTINUOUS mode.

caternuson, 2022-05-17:
OK, try changing your loop code to this:
```python
while True:
  #chan1 = AnalogIn(ads, ADS.P0)
  #print(ADS.P0)
  print(f'0. channel value is: {chan1.value}')

  #chan2 = AnalogIn(ads, ADS.P1)
  print(f'1. channel value is: {chan2.value}')

  #chan3 = AnalogIn(ads, ADS.P2)
  print(f'2. channel value is: {chan3.value}')

  #chan4 = AnalogIn(ads, ADS.P3)
  print(f'3. channel value is: {chan4.value}')

  print('\n')

  #chan1 = AnalogIn(ads, ADS.P0)
  #print(ADS.P0)
  print(f'0. channel voltage is: {chan1.voltage/voltage_divider}')

  #chan2 = AnalogIn(ads, ADS.P1)
  print(f'1. channel voltage is: {chan2.voltage/voltage_divider}')

  #chan3 = AnalogIn(ads, ADS.P2)
  print(f'2. channel voltage is: {chan3.voltage/voltage_divider}')

  #chan4 = AnalogIn(ads, ADS.P3)
  print(f'3. channel voltage is: {chan4.voltage/voltage_divider}')

  print('\n')

  time.sleep(1)
```

sgrobler, 2022-05-17:
I ran your code on my ads1115/raspberry pi but I don't seem to get the same issue.

Yingliangzhe, 2022-05-17:
The output of CONTINUOUS mode is almost the same as that of SINGLE mode.

Just because of changing the place of printing voltage?

Yingliangzhe, 2022-05-17:
> I ran your code on my ads1115/raspberry pi but I don't seem to get the same issue.

But I still have the problem, if I print the voltage just after the raw counts.

Yingliangzhe, 2022-05-17:
pip freeze tells me: 
adafruit-circuitpython-ads1x15==2.2.12
Maybe not the latest version. 

OK. 
Thank you for helping.

caternuson, 2022-05-17:
Thanks. That's latest enough. It has the latest code, for the code that matters.

@sgrobler What's your setup? Curious why you did not see the issue.

Yingliangzhe, 2022-05-18:
@caternuson 
OK. I am just a little bit admired, why the order of printing matters. 
I think if I just use the voltage of a channel should be fine. 

The Voltage_divider is for the conversion from my measurement to the real value. 
Because there are 2 resistors (15k + 10k) in a series connection. I only measure the voltage of 10k.

sgrobler, 2022-05-18:
> @sgrobler What's your setup? Curious why you did not see the issue.

@caternuson 
I have an old Rpi Model B r2 + ads1115 
adafruit-circuitpython-ads1x15 v2.2.13
adafruit-circuitpython-busdevice v5.1.8

Yingliangzhe, 2022-05-20:
Hello @caternuson,
sorry for the late repsonse. I couldn't connect to the rpi during these days. 
Now I have tested it again, following are the ouputs:

```python
CONTINUOUS

0. channel value is: 8926, channel voltage is: 2.7894601275673696
1. channel value is: 8926, channel voltage is: 2.7894601275673696
2. channel value is: 9157, channel voltage is: 2.8616498306222726
3. channel value is: 9117, channel voltage is: 2.849149449140904

0. channel value is: 8926, channel voltage is: 2.7894601275673696
1. channel value is: 8926, channel voltage is: 2.7894601275673696
2. channel value is: 8926, channel voltage is: 2.7894601275673696
3. channel value is: 8926, channel voltage is: 2.7894601275673696
```

```python

0. channel value is: 8927, channel voltage is: 2.8260237434003725
1. channel value is: 8927, channel voltage is: 2.789772637104403
2. channel value is: 9157, channel voltage is: 2.861962340159306
3. channel value is: 9116, channel voltage is: 2.849149449140904

0. channel value is: 8926, channel voltage is: 2.8260237434003725
1. channel value is: 8927, channel voltage is: 2.789772637104403
2. channel value is: 9158, channel voltage is: 2.8616498306222726
3. channel value is: 9117, channel voltage is: 2.849149449140904

```

So both modes provide almost the same output. 
Don't know what happened before. :(

caternuson, 2022-05-20:
OK, no worries. Will close this for now. If it happens again and issue can be recreated, can reopen and take another look.

closed 2022-05-20 by caternuson