adafruit/Adafruit_CircuitPython_ADS1x15 #83: Typing

scirelli-c1, 2022-08-23:
Is there a typing stub for this repo? If not would you accept a PR to add typing?

tekktrik, 2022-08-30:
Not currently, but a PR is welcome!  If you can add the typing information inline like our other libraries, that would be fantastic!

scirelli-c1, 2022-08-30:
I'll give it a shot. It might take me a little while to get the project up and running so I can test that I don't break anything :-D

scirelli, 2022-08-31:
@tekktrik Could you point me to some docs on how to setup the dev environment for this project? 

I can not find documentation on how to setup the development environment. It looks like the project users pylint (I see a lint rc) and pre-commit (I see that config file as well). I assumed the optional_requirements file was for devs but it's empty. 

If there's nothing like that, should I create a PR to add dev dependencies like pylint, and mypy (for when I  add types)

Thanks!

tekktrik, 2022-08-31:
No need to add a dependency to `mypy` as we aren't tying ourselves to it at this stage.  You should be all set for development if you can use `pre-commit` and build the `sphinx` documentation:

https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/check-your-code
https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/sharing-our-docs-on-readthedocs

scirelli, 2022-08-31:
My python version is `Python 3.10.4`
pip is at `pip 22.0.4`

Should I be using a specific version?

tekktrik, 2022-08-31:
Nope, the CI will use Python 3.10!  Do note that the minimum version supported by Blinka is Python 3.7, so note that all type annotations added must be supported by that.

closed 2022-09-07 by tekktrik