adafruit/Adafruit_CircuitPython_ADS1x15 #90: ADS1015 returning 16 bits?

jrafolsr, 2022-11-11:
Hej!

I am using the ads1015 in single ended mode and getting 16 bits when I ask for the value. If it is alledegly a 12 bits ADC shouldn't I get 4096 as the maximum value? I am missing something?

Thanks a lot,

Joan

CamDavidsonPilon, 2022-11-11:
Are you _sure_ it's an ads1015 and not a ads1115 you have?

jrafolsr, 2022-11-11:
Hej! I am pretty sure it is the ads1015, I am using the version from SparkFun Qwiic (12-bit) ADC, which only exists fot the 12 bits (and it is specified in the board).

Here is the snippet of the code I use to check the value:

```
from time import sleep
import board
import busio
from adafruit_ads1x15.analog_in import AnalogIn
import adafruit_ads1x15.ads1015 as ADS

i2c = busio.I2C(board.SCL, board.SDA)

gain = 16
address = 72
channel = 0

ads = ADS.ADS1015(i2c,gain = gain, address = address)

ch = AnalogIn(ads, 0)

while True:
    try:
        value, voltage = ch.value, ch.voltage
        
        print(f'\rvalue = {value: 6d}, voltage = {voltage:.4f} V', end = '')
        sleep(0.1)
        
        
    except KeyboardInterrupt:
        print('\nProgram terminated')
        break
    except Exception as e:
        print(e)
        print('\nAn exception ocurred.')
        break
```

thanks!

tekktrik, 2022-11-15:
The returned value gets stretched to be 16 bits even if it's coming from 12-bit resolution.  I assume that this is so it can be used as a drop in solution for onboard ADCs that also return 16 bit values.

https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/blob/1554279746863bf07b7e9f770cd9f62c33d027df/adafruit_ads1x15/analog_in.py#L53-L55

jrafolsr, 2022-11-17:
Oh! I see, I missed that, I didn't get at first this bit shifting operator.
Problem solved then. Thanks!

closed 2022-11-17 by jrafolsr

caternuson, 2022-11-17:
Just commenting to confirm the shifting is intentional and is simply a CircuitPython specific design decision. Main motivation is to have a consistent behavior for the `value` property regardless of ADC being used:
https://docs.circuitpython.org/en/latest/shared-bindings/analogio/index.html#analogio.AnalogIn.value

This is similar to Arduino's `analogRead()` returning 10 bit values by default.