adafruit/Adafruit_CircuitPython_ADS1x15 #91: Setting configuration before reading

OmerWilmo, 2023-05-03:
Hi, i'm trying to read data from ADS1015. i read the python code and it seems like there are configuration to write before reading.
i tried this code:
```
    
unsigned char buf[3] = {0};
    buf[0] = 0x01;
    buf[1] = 0xc3;
    buf[2] = 0x83;

    if (write(fd,buf, 3) != 3)  exit(-1);
```

but only one value can read and then it stops - when i'm running the python code in background it starts immediately reading data.
i guess my configuration are wrong (maybe for a single shot).
would like to get some help with the correct configuration for continuous mode. 
Thanks :)

caternuson, 2023-05-03:
Issues opened here should be related to the library code. Are you using this library somehow? The code snippet appears to be C, not python?

OmerWilmo, 2023-05-03:
Hi, i'm using this library in python. i want to do the same in c, but not sure how the config data should be written

OmerWilmo, 2023-05-03:
this is the code in python (i want to take the output of continuous mode configuration and do the same in c)

```
def _read(self, pin: Pin) -> int:
        """Perform an ADC read. Returns the signed integer result of the read."""
        # Immediately return conversion register result if in CONTINUOUS mode
        # and pin has not changed
        if self.mode == Mode.CONTINUOUS and self._last_pin_read == pin:
            return self._conversion_value(self.get_last_result(True))

        # Assign last pin read if in SINGLE mode or first sample in CONTINUOUS mode on this pin
        self._last_pin_read = pin

        # Configure ADC every time before a conversion in SINGLE mode
        # or changing channels in CONTINUOUS mode
        if self.mode == Mode.SINGLE:
            config = _ADS1X15_CONFIG_OS_SINGLE
        else:
            config = 0
        config |= (pin & 0x07) << _ADS1X15_CONFIG_MUX_OFFSET
        config |= _ADS1X15_CONFIG_GAIN[self.gain]
        config |= self.mode
        config |= self.rate_config[self.data_rate]
        config |= _ADS1X15_CONFIG_COMP_QUE_DISABLE
        self._write_register(_ADS1X15_POINTER_CONFIG, config)

        # Wait for conversion to complete
        # ADS1x1x devices settle within a single conversion cycle
        if self.mode == Mode.SINGLE:
            # Continuously poll conversion complete status bit
            while not self._conversion_complete():
                pass
        else:
            # Can't poll registers in CONTINUOUS mode
            # Wait expected time for two conversions to complete
            time.sleep(2 / self.data_rate)

        return self._conversion_value(self.get_last_result(False))

```

caternuson, 2023-05-03:
> Hi, i'm using this library in python. i want to do the same in c, but not sure how the config data should be written

Sorry, that support is not available via this forum. This is only for posting a demonstrable issues with the library code hosted in this repo.

closed 2023-05-03 by caternuson