adafruit/Adafruit_CircuitPython_ADS1x15 #92: adafruit_ads1x15.Analogin - correct ADC scaling to 16-bit range to match analogio.AnalogIn #43

kevinjwalters, 2023-07-01:
The change in https://github.com/adafruit/circuitpython/issues/4794 (raised by @PaintYourDragon) which appeared in [CircuitPython 8](https://github.com/adafruit/circuitpython/releases/tag/8.0.0) scales values to the full 16 bit range for `analogio.AnalogIn`.

The current implementation of [analog_in](https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/tree/main/adafruit_ads1x15).AnalogIn has the old behaviour for the ADS1015 of zero padding to signed 16bit for the 12bit values.

This is part of the same ecosystem, should this be brought into line with the aforementioned full-scaling change? Same issues raised in https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx/issues/43

kevinjwalters, 2023-07-01:
There's also a question here about the unsigned nature of the values. Users of analogio.AnalogIn are accumstomed to the 16 bit unsigned range between 0-65535. I've only just started reading about the TI ADS in detail but they appear to effectively be 15bit ADCs (ignoring noise) when making single ended measurements? The values align with this and the code does too

https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/blob/5ca81ee53fe296402f28758dcc778a484cc3ba32/adafruit_ads1x15/analog_in.py#L60

The `value` property will return -32767 (or -32768??) to 32767. This has a lot of potential to confuse users coming from CircuitPython's analogio.AnalogIn world unless they are using the `voltage` property.

kevinjwalters, 2023-07-02:
And for single-ended use there's a question here as to whether a floor of 0 should be applied to the values as the ADS1115 can return negative value when measuring 0V. I've experienced this myself. Unexpected negative values can cause havock.

I don't know the origins of it but, for comparison, I noticed the nRF52 code has this

https://github.com/adafruit/circuitpython/blob/9d10a3da66cd799b659457ca9785be1a33105dc6/ports/nrf/common-hal/analogio/AnalogIn.c#L124-L126

FoamyGuy, 2024-12-05:
I think maybe this was resolved in: #98 although it wasn't the primary purpose of that PR. 

There is now code here: https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/blob/a40a311b8282053cb3cc1319aa5e4b79878b1847/adafruit_ads1x15/analog_in.py#L73-L74

and here: https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/blob/a40a311b8282053cb3cc1319aa5e4b79878b1847/adafruit_ads1x15/analog_in.py#L82-L84

Which seems to handle this if I am understanding correctly. 

@kevinjwalters can you confirm whether these bits of code are resolving the original issue documented here?