adafruit/Adafruit_CircuitPython_ADS1x15 #95: I can not use it on my Rasp pi4 64bits with python3.11? what's the reason?

augustine829, 2024-04-23:
Hi,
   I am trying to run the libs, but I encounter the issue that it can not detect INA219 or the Likes issue. I am using the Rasp pi4 64bits with the python3.11 version.  but in the other same environment, python3.9, it's working. what's the reason? differency?

caternuson, 2024-04-23:
This repo is for the ADS1015 and ADS1115 ADC's. Are you using either of those? You only mention the INA219, which is something different.

augustine829, 2024-04-23:
Sure, I am using the ADS 1X15 module for ADC checking. I can not detect it when I run my code.

caternuson, 2024-04-23:
If you run the basic example from this library:
https://github.com/adafruit/Adafruit_CircuitPython_ADS1x15/blob/main/examples/ads1x15_simpletest.py
what output are you seeing in the serial output?

augustine829, 2024-04-23:
The output from my terminal here:
 raw	    v
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002
   16	0.002

caternuson, 2024-04-23:
It's working, so it looks like it's detecting the ADS? It's unclear what the issue is?

augustine829, 2024-04-23:
No, it can not initialize the ADS module, but I checked the voltage from the ADC pins with a multimeter.

augustine829, 2024-04-23:
ok, thanks. let me do it now.

augustine829, 2024-04-23:
I check my address 0x49 with channel-0, it outputs here:
=============================================
 raw	    v
  275	0.034
  276	0.034
  276	0.034
  276	0.034
  275	0.034
  273	0.034
  278	0.035
  279	0.035
  269	0.034
  274	0.034
  276	0.035
  273	0.034
  269	0.034
  276	0.034
  276	0.034
  275	0.035
  265	0.034
  277	0.035
  276	0.034
  273	0.035

FoamyGuy, 2024-11-05:
Closing this for now, as there was no further response. Please feel free to re-open or create a new issue if you are still having any issues.

closed 2024-11-05 by FoamyGuy