adafruit/Adafruit_CircuitPython_ADS1x15 #96: error in ads1015.py

fzdm2022, 2024-06-17:
Traceback (most recent call last):
  File "/Users/nianhong/kairos/kairos_repo/vibration_monitor/ADS1015_FT232H.py", line 10, in <module>
    import adafruit_ads1x15.ads1015 as ADS
  File "/Users/nianhong/kairos/kairos_repo/vibration_monitor/venv/lib/python3.11/site-packages/adafruit_ads1x15/ads1015.py", line 23, in <module>
    from .ads1x15 import ADS1x15, Mode
ModuleNotFoundError: No module named 'adafruit_ads1x15.ads1x15'

caternuson, 2024-06-18:
It looks like you are using Python with Blinka via a FT232H.

How did you install the ADS1x15 library on to your system?

FoamyGuy, 2024-11-05:
Closing this for now since there was no further response. please feel free to re-open or create a new issue if you get back to this and are still having issues.

closed 2024-11-05 by FoamyGuy